robots/gdm #2: Revive old GDM610

altenplos, 2021-07-17:
Hi there, that's a great job you've done!
I just saved an old GDM610 from being thrown away but its onboard batteries are dead -> program lost ...
Is there by chance any possibility to bring this loveley device back to live with these tools? The serial port seems to work fine on the CU

robots, 2021-07-17:
Hi,
when battery dies in gdm610 all content of memory is erased: that means firmware, calibration values, etc etc. You can bring it up, by uploading firmware from other unit (preferably very similar one - same tracker edm version) and it will just work. Although with uploaded firmware calibration is not the same. You wont get the same accuracy as before. This is the trick that geodimeter used to make you come back to them and now that the stations are not supported they are thrown away.

I am working on calibration routine for 400/500/4000 stations. As the sensors are the same, and gdm6x0 has servo board it "should"  be easier to calibrate, but this is very far away. I need to find how to control servo board not by angles but by "steps". Too much work to be able to give you date.

I have saved also one GDM600 from dump, and i have uploaded firmware from different gdm600. It took many days to find out how to change the configuration of the station (mine is without tracker), now station works, but CU will not connect. needs more work

altenplos, 2021-07-19:
Maybe EPROMs didn't make their way to sweden back in the 90s.. :)
I'm afraid there is no similar unit I could use, would you mind sending me one of your dumps? Just to test if things are working...

Jesse-EE, 2024-04-21:
> Hi there, that's a great job you've done! I just saved an old GDM610 from being thrown away but its onboard batteries are dead -> program lost ... Is there by chance any possibility to bring this loveley device back to live with these tools? The serial port seems to work fine on the CU

I have a couple working GDM610's I could send the FW dump from. They both have EDM, tracklight, and will track the RMTs. I don't know if it'll work on yours but if you like trying things....

altenplos, 2024-04-24:
> I have a couple working GDM610's I could send the FW dump from. They both have EDM, tracklight, and radios , and will track the RMTs. I don't know if it'll work on yours but if you like trying things....

Thank you for the offer, but I don't have the 610 any more. I bought an old Trimble 5600 without CU :(

Jesse-EE, 2024-04-24:
> Thank you for the offer, but I don't have the 610 any more. I bought an old Trimble 5600 without CU :(

Do you know if your 5600 works, or is it battery-brain-dead?

I also have a Trimble 5601 DR200+ which is battery-brain-dead -- if you can read the code from yours I'd love to try it on mine.

I don't have a CU for mine but I do have a CU for a 610, and I have a TSCe Data Ranger / Collector that should work, so if I could get code in my 5600 and get it to respond to the  Data Collector, I should be able to capture the RS232 commands so you could make your own CU or control it from a computer or whatever.

altenplos, 2024-04-25:
> Actually, I suspect all the commands are documented for the serial port so if your firmware is still alive in your 5600 you should be able to control it via RS232 and take measurements.

Unlike other manufacturers (Leica, Sokkia, Topcon from which i own a TS) GDM/Trimble has only limited documentation for the RS232 interface of their older devices. I couldn't find out how a EDM measurement can be triggered on a 5600.
For the FW of the 5600 you can pm me tr2@fregga.de

Jesse-EE, 2024-04-25:
> Unlike other manufacturers (Leica, Sokkia, Topcon from which i own a TS) GDM/Trimble has only limited documentation for the RS232 interface of their older devices. I couldn't find out how a EDM measurement can be triggered on a 5600. For the FW of the 5600 you can pm me [tr2@fregga.de](mailto:tr2@fregga.de)

After doing some net searching, I agree with you. I was unable to find anything other than a few sample commands.

I was able to trigger EDM on a GDM620 Pro via RS232.

Basically, TG triggers the EDM sequence, but returns no numbers.
You then do RG,9 and it then gives me the distance in meters.

Alternatively, you can just do RG and it gives both angular readings and the last EDM reading.

However, you can only read out the EDM reading once. After that, it gives error 3 "Already read."

You have to do a TG again to re-activate the EDM then you can RG it again.

I have no idea if this is similar to the 5600 or not. I'll shoot you an email note!

I do have a TSCe which does work with ONE of my 610's but not both... 

I have parts on the way to build an RS232 debugger, hopefully I'll soon have some data dumps of the communications between the Data Collector and the 620/610. (And if I can get your 5600 code communicating on my 5600, I should be able to data dump on the RS232 on that too.)

Jesse-EE, 2024-04-26:
>  I couldn't find out how a EDM measurement can be triggered on a 5600. For the FW of the 5600 you can pm me [tr2@fregga.de](mailto:tr2@fregga.de)

I just loaded your 5605 firmware into my dead-battery-blank 5601!

It took it, and I was able to type TG to trigger an edm, then RG to read it and the two angles.

I was also able to connect my TSCe Ranger Data Collector and do some stuff with that, including angles and controlling tracklight and stuff. Even tracks my active RMT. 

However, the TSCe so far can't trigger an EDM function so I don't know what's with that. But it triggered and read via RS232 as I mentioned with TG and RG.

Not surprising, the TSCe Ranger now identifies my 5601 as a 5605.

But the initialization procedure worked - the Bubble level display came up on the TSCe Ranger and  let me level it out, did the double-half-rotation initialization of the level compensator just like normal.

Of course my ram battery is still dead so if it powers off I have to send in your firmware again LOL.

LATER IN THE DAY:

I replaced the ram battery. Now my 5601 identifies as a 5605 and sort of works, but it has a 2 minute timeout on power up during which it does not respond to RS232 or CU. After that it comes alive and does some stuff. It's like it's trying to connect to some internal feature? Laser?) that mine doesn't have, and it gives up after 2 minutes.

Sometimes I can get it to do the EDM, sometimes I can't. I don't yet know exactly what the deal is. (It is from eBay, it could be faulty. But the EDM does seem to be working, I think maybe the firmware just isn't quite right. haha!!)

I have some parts on order to build a bi-directional RS232 recorder so I can hopefully get some two way data stream dumps between the TSCe/Ranger and the 5600.

robots, 2024-04-28:
BTW: 5600 station and 600 station are the same. (600 with PVX2 board, there is also 600 with older PVX board, and this firmware is not compatible with 5600)
One exception is the EDM, 5600 stations use different EDMs - DR, DR200 and DR300. But the firmware is the same - different configuration.

Jesse-EE, 2024-04-28:
> BTW: 5600 station and 600 station are the same. (600 with PVX2 board, there is also 600 with older PVX board, and this firmware is not compatible with 5600) One exception is the EDM, 5600 stations use different EDMs - DR, DR200 and DR300. But the firmware is the same - different configuration.

Interesting! So I could try loading my 5601 with the firmware from my two 610's and  my 620. I think I did try the firmware from one of the 610's, but it just beeped and beeped and beeped.

I did put that other fellow's 5605 code on mine, and the EDM is hit or miss (mostly miss) and I cannot zero out the vertical index error -- in other words, F1 and F2 are different by about 1.3 degrees.  

However, if I take F1 and F2 and average them the average is spot on.

Is this likely due to slight changes in hardware that was compensated for in the firmware? 
I did try physically rotating the vertical angle sensor but was not able to get it below 1 degree of difference.

MattMalone, 2025-01-17:
I am looking to revive an old GDM 608 S -- I thought it was similar enough to a 610 not to warrant a new thread.

OK, I am reading over the repository, and, like all new-to-me equipment without a technical service manual, I am missing the basic introductory steps, knowing what I do not know.   A few direct answers would greatly assist me in doubting the basics:

Q1) Its it true that the GDM608 body has firmware, and separately the GDM608 CU has firmware ?
Q2) Is it true that I access/download/upload the GDM608 body firmware by plugging into the port on the tribrach ?
Q3) Is it true that I access/download/upload the GDM608 CU firmware by plugging into the port on the right side of the CU ?

The current state of my system is, the CU powers and enters Local Mode when attached to the CGM608 body, when powered through the tribrach.   Whenever asked to do anything with the GDM608 body, the CU says No GDM Connected.  The body responds to the horizontal motion servo control knob and the vertical motion servo control knob by panning and tilting the telescope with a type of knob-motion response that looks like software to me, not an analog circuit, but I could be wrong.   The GDM system draws about 0.75A when it is stationary, and this goes to 0.9A when moving.   

Q4) Does the response of the unit to the motion servo control knobs on the side indicate that the firmware in the GDM608 body is intact, or could this firmware be lost ?

Q5) Does the CU lighting up and going into local mode when powered from the tribrach indicate that the firmware in the GDM608 CU is intact ?

Q6) Does the CU entering Local Mode and not recognizing the GDM608 is present indicate that the problem is a communication problem between the GDM608 and CU ?   I have already cleaned the pins (they were pristine before I touched them) and I have slightly bent the spring-loaded contacts on the GDM608 body to make better contact to the back of the CU -- no improvement.   If you think the problem is still communications, any other ideas?

I do not see this file in your repository:

https://geodesical.com/es/pdfs/articulos/geodimeter-catalogo-estacion-total-serie-600-en.pdf 

Which gives an overview of the specs and configuration options of the GDM600 series.

My unit is a GDM608 with "Overview of upgrades and options" of:

1 - standard side cover
6 - tracker
10 - Numeric Control Unit
20 - RMT - Long Range model

Q7) I have looked under every cover, I can find no evidence of a battery to power the unit, though I found two flat lithium cells in the CU and two flat lithium cells in an inaccessible location on a board in the GDM608 body.   Is is true that in this configuration (options 1,6,10, 20) that there is no internal battery to power the unit ?

Q8) I measured one of the two flat lithium cells in the CU to be 2.85V, the other was 3.2V.   I intend to round the firmware download code on the CU prior to dealing with the low cell.   Are these voltages incompatible with my belief that firmware is still present on the CU ?

Q9) Other than what I have done, what is the easiest test to know if my GDM608 body still has live firmware ? 

Q10) I have a generic USB to RS-232 cable but I am uncertain how my linux system will recognize it, or the GDM if I connect it.   I have decades of experience with RS-232 hardware connected to DOS hardware, and used USB hardware connected to modern hardware, but this is my first attempt to use RS-232 hardware connected to a modern computer.   I have not other legacy RS-232 hardware (threw away all my old modems) I do not care about to run tests.   Can someone paste some keywords in here that I can google, or give some tips to help make sure the connection goes smoothly ? 

Q11) Last question, promise.   Is the shop service manual for any 600 series GDM available anywhere ?  Something that would walk a person through procedures, identify internal boards, describe diagnostics ?

Equipment and Current Test Setup
----------------------------------------------
I have only one Hirose to Hirose (H-H) cable, and one Hirose to RS-232 cable.   There is also an external battery interface module with two bare battery contacts and two Hirose connectors.  I used the Hirose cable and the functioning RMT to give me information on polarity of the two bare battery contacts. I have used alligator leads to feed 13.5V into the battery connection.    The 13.5V comes from a laptop power supply feeding a buck/boost converter set to 13.5V, current limited (CC) at 1A, to a 75kmicroF cap to smooth it out and imitate an SLA battery (nothing happened without the cap), with the H-H cable from the external battery interface module into the tribrach-level connection on the GDM608.

bdm310, 2025-01-18:
I've been collecting a lot of my findings on a 608s in this discussion thread: https://github.com/robots/gdm/discussions/11
Robots has reasonable copyright concerns about sharing binaries so I won't link here but I have a repo of firmware images for reverse engineering that you can find.

Q1 - Yes.
Q2 - Yes. Along with some other steps.
Q3 - I don't believe extraction of the CU firmware is possible by default. Robots has a method that isn't documented and I think requires some custom software that hasn't been published to this repo yet.
Q4 - I'm not quite sure. There's about 5 total processors in the unit depending on your options. The servo board has it's own 8051 with firmware in EPROM so it may just work even if the main board firmware is gone.
Q5, Q6 - I believe this indicates that your main board firmware is gone. If both of the coin cells inside the unit were dead then that is certainly the case.
Q7 - You have the tracker option, which probably means there is no place to put a battery on the unit. On mine, the battery goes in where the tracker would go if I had it. Probably you have to power through the tribrach.
Q8 - Seems likely that your CU has firmware, especially if it turns on and gives you messages on the screen. Those battery voltages are fine, though I'd suggest replacing them starting with the 2.8V one first. Very carefully, without shorting anything!
Q9 - I'd suggest reading the linked discussion thread, if you can get a console output the way I describe then your main board firmware is intact and you can extract it.
Q10 - I've been using a generic USB-RS232 adapter successfully with the Hirose-RS232 cable that came with my unit.
Q11 - Robots has the best repository I can find of documents.

MattMalone, 2025-01-18:
I am most grateful for your direct answers that clarify many things ... multiple processors in the GDM608 body... this only makes sense.  I will read and reply further in the other thread.

Jesse-EE, 2025-01-18:
> I am most grateful for your direct answers that clarify many things ... multiple processors in the GDM608 body... this only makes sense. I will read and reply further in the other thread.

Hey Matt, In the past I've messed around with a 5601 and a 610 and a 620.
The 5601 came to me with dead ram batteries and was dead dead dead except the knobs actuated the servos.
Someone on this forum sent me their 5605 firmware and I was able to use Robot's upload.py tools in linux with USB-RS232 adapter to put the 5605 firmware onto my 5601 -- which brought it more or less back to life (but some of the options are different so the Direct Reflect doesn't seem to work but basic functionality works.)

I have also downloaded the firmware from the two 610's and the 620 which I'll be glad to send you if you want to see if you can stuff it in there and at least get signs of life (however obviously all the calibration tables will be for my units, not for yours.... But reasonable chance at least it will act like it's working!)

I tried to attach to here but would not work.

And you are correct, the CU goes into local mode if it cannot connect to station. That's because it does have the ability to store data on itself, so you can just take the CU into the office and load data on/off rather taking  the whole total station to your desk.

MattMalone, 2025-02-02:
@Jesse-EE 

> I have also downloaded the firmware from the two 610's and the 620 which I'll be glad to send you if you want to see if you can stuff it in there and at least get signs of life (however obviously [...]

May I get a copy of each of your two 610 downloads please ?

Jesse-EE, 2025-02-02:
> [@Jesse-EE](https://github.com/Jesse-EE)

> May I get a copy of each of your two 610 downloads please ?

They should be here: https://github.com/bdm310/Total-Station-Firmware/tree/main/main/binaries