robots/gdm #3: ValueError: invalid literal for int() with base 16: '0 0'

Tecnosoft-dev, 2021-09-02:
hi robots

when running backupfirmeware.py it starts reading normally until 28% shows this error:

[image]

the instument that i m reading from is working, i have tested also with another working station and the same error is shown,

i m using python 3.4.1 installed on win7 pack 1, 32 bit arch.

as i have a few experience in programing i think the error is because the int with base 16 cant quary more data

so what is the simple solution

Tecnosoft-dev, 2021-09-02:
https://stackoverflow.com/questions/11826054/valueerror-invalid-literal-for-int-with-base-16-x0e-xa3-python

may this help?

Tecnosoft-dev, 2021-09-02:
[image]

this screen shot is more clear hope help you solving

robots, 2021-09-03:
Hi, the data coming from station looks wrong.
It should be 16 pairs of numbers, not triplets as in your case. I would suggest trying with more recent python version and pyserial. Python 3.4 is very old one. I work with 3.9.

Tecnosoft-dev, 2021-09-03:
I installed the most recent python version 32 bit on windows 10, and tried again the same error apears, but now now it ends in 7 % or 23 % of reading not 28%.

This is the header of the printed log:
[image]

Tecnosoft-dev, 2021-09-03:
This what is being in buttom of one of my two  instrument:
[image]

To know i have tried dumping three models of instrument but the same error apears, one of them working perfectly 5605, the others 5605 and 5601 not working:
5605: we have changed the two internal batteries missing bad side effects like loosing internal program, now it is not connecting and not detected at all by the CU "please connect the instrument..." But it runs with it self.

5601: the cu tells "unsupported instrument model", and says also "internal batteries expired"
But still not functioning.

Tecnosoft-dev, 2021-09-04:
I tried to modify the code where the error appears, so i became with this code:
```
                z = data[3*i:3*i+3]
                if len(z) == 3:
                  k = z.replace(" ", "")
                  if len(k) != 0:
                    if len(k) == 1:
                      bb = int(k,16)
                    elif len(k) == 2:
                      e0 = z.split(' ')[0]
                      e1 = z.split(' ')[1]
                      bb = int(e0+e1,16)
                    mem_main[addr+i] = bb
                    continue

                mem_main[addr+i] = int(z,16)
```
as shown here:
[image]

so it reads until 53% then stops with same error but with another string as shown in image:
[image]

Tecnosoft-dev, 2021-09-04:
i also tried adding this code:
```
                elif len(z) == 2:
                  if z == '\r\n':
                      continue
```
before this:
`mem_main[addr+i] = int(z,16)`

so it completed mem reading then it started reading bank until 19% then almost the same error appeared:

[image]

Tecnosoft-dev, 2021-09-04:
may this help:

https://stackoverflow.com/questions/44241453/python-valueerror-too-many-values-to-unpack-expected-2

Tecnosoft-dev, 2021-09-04:
i also tried updating pyserial to the most recent version by following this video:

https://www.youtube.com/watch?v=Pf-cGzOQmXU

but still the same error uisng your unmodified code

Tecnosoft-dev, 2021-09-04:
[image]

I'm trying with this new suggested code so it may solve the bank reading error "too many ..."

Tecnosoft-dev, 2021-09-04:
Now its showing another error when is in bank reading 21%:
[image]

I think i haven't touched the problem's head yet, i think i have the same instrument of yours.

Tecnosoft-dev, 2021-09-04:
If you don't mind can you send me your extracted firmware file so i directly upload it to 5605 or 5601 or 5603.

Tecnosoft-dev, 2021-09-04:
i then modified the code:

```
                zzz = data[3*i:3*i+3]
                if len(zzz) == 3:
                  k = zzz.replace(" ", "")
                  if len(k) != 0:
                    if len(k) == 1:
                      bb = int(k,16)
                    elif len(k) == 2:
                      e0 = zzz.split(' ')[0]
                      e1 = zzz.split(' ')[1]
                      bb = int(e0+e1,16)
                    banks[bank][addr+i] = bb
                    continue
                elif len(zzz) == 2:
                  if z == '\r\n':
                      banks[bank][addr+i] = 0
                      continue
                    
                bb = int(zzz,16) 
                if bb < 256:
                   if bb >= 0:
                      banks[bank][addr+i] = bb
                      continue
```
the progress ends in bank reading after about 6 minutes, bank 5/8 ... 70% and show the same error:

[image]

Tecnosoft-dev, 2021-09-04:
Is there any different between running the code in python program and running it in black command console of windows,
I mean showing those errors.

Tecnosoft-dev, 2021-09-05:
I used cmd to run tha prog so everything worked as wanted.

AkramBsn, 2021-09-05:
sir can you send me firmware for 5603 dr200+ please if possible i have problem with mine not working for me  :/ ??

robots, 2021-09-06:
@Tecnosoft-dev  so the dump worked ?

Tecnosoft-dev, 2021-09-06:
yah, worked just for 5605 but not with 5601,
5601 i dumped the memory into a life then  i replaced the old two internal batteries then i re-upload the file forcingly (skipping errors) but in the end it still showing "unsupported instrument model, please contact..." despite the "internal battery expired" is really solved.

Tecnosoft-dev, 2021-09-07:
**i posted here for 5601 problem:**
[https://surveyorconnect.com/community/surveying-geomatics/trimble-5601-unsupported-instrument-model/](url)

robots, 2021-09-09:
> 
> 5601 i dumped the memory into a file then i replaced the old two internal batteries then i re-upload the file forcingly (skipping errors) but in the end it still showing "unsupported instrument model, please contact..." despite the "internal battery expired" is really solved.

What does it mean "skipping"errors ?

> 
> so i need a another healthy dump for 5601 and 5603, if you have the two or at least one please send it to this mail:
> 

Its not that easy, Its not just firmware you are uploading but also configuration of station and calibration. These stations come in many different configurations - different EDM modules, radio, tracker, etc. Calibration values are valid only for the angle sensors in the station and values from different station will work but will provide you wrong readout.

closed 2021-09-09 by robots

Tecnosoft-dev, 2021-09-09:
Skipping errors: putting # before sys.exit and some other lines, because sometimes it doesn't upload normally.
About values: because uploading a working firmware remain as one of resting probabilities i could test, even if values may deffer, so if it works i believe the issue is not in hardware but in software. Like that i can move to next step, which is looking at adjusting values.

Tecnosoft-dev, 2021-09-09:
You may know that before:
[image]

I converted the firmware resulted using a site hex to ascii.

Tecnosoft-dev, 2021-09-09:
When i compared two firmwares one of working station and one of not working, i see that all the difference is in the head of the two files (from 0 to about 310 lines), the rest lines are exactly the same (from about 310 to the last).
May this help!

robots, 2021-09-09:
Yes, the begining is the working ram and data. Some of the data is working buffers for serial port, and some is the calibration data.

Tecnosoft-dev, 2021-09-09:
You may noticed (from Michelle 's papers he captured in survey connect forum) they used a program named: HexLoader, and by searching i founded a copy in GitHub.

I made the previous comparison by compare plugin of notepad++, its useful.

I actually downloaded the firmwares of 5601 and 5603 before reinstalling the two new 3V batteries.
But as i told you before i uploaded the firmwares and yes the stations revived (no more battery messages) but "unsupported instrument model" and "detected but not connecting" still the same problems.

So by converting the a file from hex into ascii (image above), i see a lot of values i fairly believe they are the values mentioned in Michelle's papers.

So to put you in the image, i just came with an idea of making somehow a new file contains the same program but only the values will be changed (depends on the station uniques), but a problem of number of lines will came up, so to solve this:
1. We need a working firmware of each station model 
2. A program that analysis the old firmware and extract all values included.
3. Another program merges between the healthy firmware and the old values of the station.
4. Upload the new firmware file.

robots, 2021-09-09:
As i said, its not that easy :-) 

what you are seeing .. those texts ... are actually serial port buffers. Data that was sent over serial port or received over serial port. Label values are in binary. You can actually modify those, almost any of them, over serial port! After you login into station and unlock those labels.

Problem is, that you don't know which label you need to change to what value.

Tecnosoft-dev, 2021-09-09:
Would you please tell me how to change them, how to convert them to human readable text and return them back to hex buffers.
Because "unsupported instrument model" message apears and don't let cu to use station, for that i have a probability the old firmware was corrupted before i downloaded it, so i think changing 'model' value may solve the problem

robots, 2021-09-10:
You can access all labels from serial port of station. There is no need to access them in hex editor and change them. Problem is that, all labels are numbers and those numbers are not very well documented.

Tecnosoft-dev, 2021-09-10:
Would you please shiw me an example of code on how to read a value and write it.

Tecnosoft-dev, 2021-09-11:
Why gdm_write can't change value 100 (gdm model) it just results: 20.8, when i read the value it still the same.

robots, 2021-09-18:
https://github.com/robots/gdm/tree/master/docu/workshopscan
here .. have fun with labels :-)
to read label you need to use Rx,nn ... where x is the "subsystem name", S- servo board, G - gdm, A angle board, etc, nn is the label number
to Write .. Wx,nn=value
all commands are sent over serial port. Some labels are locked, you need to unlock it (as the script does in the beginning - seed/key)