robots/gdm #4: upload 100% but the gun the working

AkramBsn, 2021-09-05:
Hi, first i wanna thank you a lot for this fantastic work. so i have a 5603 dr200+ that was showing a message that the internal battery has been expired and i need to change it then the gun stopped connecting, it was flashing back and forth at level bubble, so i used the backup programme to get the frimware from the gun it worked 100% i got a file with 466KB then i did changes batteries, i uploaded the FW.s file it worked also and showed 100% then the gun did a sound, then the gun still not connecting to the cu and neither the backup programme or the upload programme working what can be the problem are the firmware from the station currupted because the batteries are almoste died or it could be something else !!? any help is so apreciated

robots, 2021-09-06:
Hello, as the station turns on, the firmware is not corrupt. There is checksum that would prevent that.

AkramBsn, 2021-09-06:
is there a way to clear the memory  and reupload it without removing batteries ??

robots, 2021-09-19:
there is soo much left to discover in the firmware :) ...

there is command that will wipe all "info" from the station memory, but not the firmware itself. But i don't know how to put the data back in this state.