robots/gdm #5: USB driver/conectivity

IanR27, 2022-01-10:
Hi Robots,
A great piece of reverse engineering.
I am playing with a 400 series unit, which has the error 51 issue, I have replaced the memory backup battery, built the interface board, flashed / verified the STM32 device. My problem is the STM32F103 is not recognised when I connect it to ubuntu or windows via USB.
It appears no ST Linux drivers have been drivers have been published for the part, please could you advise how you are connecting to the device ?
Thanks.

robots, 2022-01-10:
Hi, i didnt think someone will try to replicate this :-)

are you using bluepill stm32 board ?  Or is that some other board ? bluepill has built-in always on USB pullup. If your board is different, this might be problem.
Otherwise the USB acts as ttyACM so no linux driver is necessary.

IanR27, 2022-01-10:
Yes, as far as I can determine a genuine STM32 Blue pill, as per your warning. Tonight I have tried 4 different variants, and none have been recognised as a USB device. I have just noticed, the circuit diagram of the blue pill shows a pull up resistor of 10K, but the datasheet says 1k5, and no 15k load resistors are shown. Not used a blue pill USB before....

robots, 2022-01-11:
My bluepill is definitely genuine STM32. There is 22ohm resistor in series
with Usb lines and 1.5k resistor pullup. R9-11 in picture.
There were some fake stm32 pills with all peripherals working, but
different usb.
Does the led blink?

IanR27, 2022-01-11:
Hi, I checked the resistors with decent probes, and all is correct. I have programmed up 5 different boards, none will connect to the USB, and no flashing LED.
Q. I am not connected to a Geodimeter at the moment, is this required for normal operation ?
Thanks,
Ian R.

robots, 2022-01-11:
Geodimeter is not needed. You need to build the gdm firmware. To make it, you run "make" in gdm-iface directory, and you need gnuarm toolchain. I use version "9 2019 q4", its fairly old, but its stable for my needs. 
After loading the firmware (stm32flash or stlink, this choise is up to you) the led should blink! short short long or similar pattern. If not, the firmware is not running. 
(I suspect your board is not genuine stm32, can you post picture of the mcu chip? some nice resolution)

IanR27, 2022-01-28:
Hi Robots, 
Still playing with the 400 series GDM, I have verified that I can flash the STM32 devices correctly, so its down to the blue.bin file, having a think, I recall the serial source file was missing from your original code drop, I assumed this was a generic file, found an STM32 serial.c  source file, copied it in to the build folder, and everything compiled ok. Now I am pondering, did you have a modified serial source file ?  I am in the process of installing "9 2019 q4" as suggested, but that is proving to be another learning curve.
Best wishes,
Ian R.

robots, 2022-02-23:
I have not forgotten about this issue :-) I am working on new release, with 2 usb ports for windows, with INF driver and separate VID/PID for usb.

robots, 2022-02-23:
> Hi Robots, Still playing with the 400 series GDM, I have verified that I can flash the STM32 devices correctly, so its down to the blue.bin file, having a think, I recall the serial source file was [...]

I have no idea what files were missing. There is not serial.c in whole project :-).

Still, i have added binary file "blue.bin" that is the very latest one. I have just tested it on my board. Try it.

closed 2023-06-30 by robots