robots/gdm #7: 5503 CU  fw file

AlexYang30, 2022-04-02:
Hi robots,
I tried to replace an internal battery of my 5503 CU,
then CU did not turn on ......
pwn + con , no luck , say "load default ..unsuccess"

so would you please send me a copy of fw.bin to me 
Thanks

robots, 2022-04-03:
Was the CU working before battery update ?

AlexYang30, 2022-04-03:
yes

從 Windows 的郵件<https://go.microsoft.com/fwlink/?LinkId=550986>傳送

寄件者: ***@***.***>
傳送時間: 2022年4月3日 下午 01:52
收件者: ***@***.***>
副本: ***@***.***>; ***@***.***>
主旨: Re: [robots/gdm] 5503 CU fw file (Issue #7)

Was the CU working before battery update ?

AlexYang30, 2022-04-04:
yes

AlexYang30, 2022-05-25:
Hi,
my email : alexyangtw@hotmail.com
Is that possible to ask some questions in private ?

Middim, 2022-07-06:
I have the same situation with missing firmware. Was this ever figured out?

Jesse-EE, 2024-04-23:
> Was the CU working before battery update ?

Does the CU also strictly use battery backed RAM or does it use any sort of flash memory?