robots/gdm #9: Differences between two backups

glasscake, 2023-06-25:
Contex:
Instrument: Spectra Focus 10
backups are done on the same boot and serial connection
So the two backups have discrepancies between them only in the initial couple of lines. Any idea why this might be?
after this section both files are identical.

Are these good backups? would it be safe to change the batteries?

Initial
S1130020002801300111E600065C0101116D000099
S1130030000000000005B5000D520112FE0111E19F
S11300400111DE11E6024CB30000000000000001C3
S113005011E301116B003DFF0001116C000000FB76
S11300600042030247C6000000032D0001072201DD
S113007002020111E4009F000000000000010200E0
S1130080000000000000011EB8024E4900F00257B3
S113009043067F025743067F026CD3026C002A0199
S11300A01DFC04026C82C000F0010000000000008E
S11300B03E000001020000000000000000000000FB
S11300C00111E50000000000000000000000000035
S11300E0000000000000000000000000000100FC0F
S11300F00AFFFF010D60000000302E3000373D374D
S11301003836310033363035003435003239323840
S1130110320000003246373030000000000000009A
S11301F000000000000000000000010D600000008D
S11302000D00040111E10D000111DE01000115448E
S113021007010BEA0BEA024D85010BEA000000001E
S1130220000000000000000000000007DB031300D2
S1130230000025785807DB03130712010714010790
S11302401500020064000001000000130111EB001E
S11302500000000000000000000001026801026BC1
S113026001026301027400020111F0024E6A0102EC
S113027082010286010C0000000000000000000062
S113028004FF00000015000000130111EB010294AB
S113029007E7050A04FF000111E4000000007200F2
S11302A072720000000001037E0002E70102F00008
S11302B0000007000000008BA98AC600060102E8BE
S11302C000000000FFFFFFFF00000003000000002B
S11302D000000000000000000A0000000001036CA0
S11302E001036C024EAA0047302E303030303030DB
S11302F03938000000000000000000000000000089
S11303600000000000003032373836300002489C6C
S11303700550E301037E0103790103860003010DA7
S1130380630248970000007F000000003AB5BA8875
S11303A00000000000C2000112E301116801117491
S11303B00105AE0000000103D700000000478000E3
S11303C0000000000000000000000000000450DBFA
S11303D0067F0111E50450DB00000000000000006E
S11303E0000000000000000000000000090105A654
S11303F00001018356080328011C3101058FFF0108
S113040002000112DE412C3136353D3338372E33AC
S1130410333630350000000000000000000000000A
S113050000000000000138372E3333363035000048
S113051000000000000007025604005CDBDB560408
S1130520005E30A200000000010112E40201053364
S113053001053743C7FFE642ACAF7000000000423C
S1130540CD4F70000111E500000000000002574388
S11305500450DB067F0112E443BCA41F000111E533
S11305600001000111E5000000570111E5570111D8

subsequent
S1130020000102300111E600065C0101116D0000BF
S1130030000000000005B5000D520112FE0111E19F
S11300400111DE11E6024CB30000000000000001C3
S113005011E301116B003DFF0001116C000000FB76
S11300600042030247C6000000032D0001072201DD
S113007002020111E4009F000000000000010200E0
S1130080000000000000011EB8024E4900F00257B3
S113009043067F025743067F026CD3026C002A0199
S11300A01DFC04026C82C000F0010000000000008E
S11300B03E000001020000000000000000000000FB
S11300C00111E50000000000000000000000000035
S11300E0000000000000000000000000000100FC0F
S11300F00AFFFF010D60000000302E30003231385D
S11301003332003333363035003435003239323847
S1130110320000003246373030000000000000009A
S11301F000000000000000000000010D600000008D
S11302000D00040111E10D000111DE01000115448E
S113021007010BEA0BEA024D85010BEA000000001E
S1130220000000000000000000000007DB031300D2
S1130230000025785807DB03130712010714010790
S11302401500020064000001000000130111EB001E
S11302500000000000000000000001026801026BC1
S113026001026301027400020111F0024E6A0102EC
S113027082010286010C0000000000000000000062
S113028004FF00000015000000130111EB010294AB
S113029007E7050A04FF000111E4000000007200F2
S11302A072720000000001037E0002E70102F00008
S11302B0000007000000008BA98AC600060102E8BE
S11302C000000000FFFFFFFF00000003000000002B
S11302D000000000000000000A0000000001036CA0
S11302E001036C024EAA0041302E303030303030E1
S11302F03938000000000000000000000000000089
S11303600000000000003032313833300002489C75
S11303700550E301037E0103790103860003010DA7
S11303806302489700000000000000003AB5BA88F4
S11303A00000000000C2000112E301116801117491
S11303B00105AE0000000103D700000000478000E3
S11303C0000000000000000000000000000450DBFA
S11303D0067F0111E50450DB00000000000000006E
S11303E0000000000000000000000000090105A654
S11303F00001018356080328011C3101058FFF0108
S113040002000112DE412C3136353D3338372E33AC
S1130410333630350000000000000000000000000A
S113050000000000000138372E3333363035000048
S113051000000000000007025604005CDBDB560408
S1130520005E30A200000000010112E40201053364
S113053001053743C7FFE642ACAF7000000000423C
S1130540CD4F70000111E500000000000002574388
S11305500450DB067F0112E243BCA41F000111E535

robots, 2023-06-27:
Its the content of ram, that contains:
- calibration data for angle sensors and servo board
- what boards are present in the station
- other settings.

Very beginning is actually used as serial buffer, so you might find some commands you used on the station recently.

glasscake, 2023-06-28:
### Very beginning is actually used as a serial buffer, so you might find some commands you used on the station recently.

this makes sense, i was thinking it might be the clock or some other nonimportant info.
BTW did the guy who made the calibration .stl ever follow up on how to actually perform the calibration? I'd love to start compiling all of this information into some forum post because these machines are an amazing option for student/hobbyis surveying.

closed 2023-06-30 by glasscake

robots, 2023-06-30:
the guy that made those stls is me :-). And this repository is compilation of all information. If you want to contribute you are very welcome to join. Clone the repository, make change and make pullrequest to merge the change.