robots/gdm #10: Trimble 5600 how to trigger a measurement

altenplos, 2023-12-02:
Hi, I just bought a 5605 station without cu and want to operate it via serial communication. 'RG,7' and 'RG,8' are giving results, but when sending 'TG' then nothing happens. Do you know the correct commands to start a distance measurement?

Btw I have this old geodatwin software with 'drivers' for geodimeter (although I can't get it to work), maybe this could be interesting for you...
My mail: joerg.wu@t-online.de

robots, 2023-12-02:
This is more complex, and i don't remember all the commands.

Rx,n commands read label "n" from unit "x".
Wx,n command writes label "x" to unit "x".
And there are also Px,n commands. These run sequence of "commands".

The compensator init and measurement is some of these Px sequences.

The main research so far was around internal working of the firmware - how to make it work after ram is erased, how the communication between internal boards is working and how to communicate with 400/500/4400 station (these have almost same hardware). What data is sent to angle board, internal working of angle board and servo board.

What i have in pipeline is 600 CU emulator. Do not expect fancy application. It already works on my computer (linux) and the serial communication is not yet finished. But i dont have access to 600 station now (and wont have it for another few months)
Best way to do this is to sniff the communication of CU and STATION and figure out what they do.

altenplos, 2023-12-02:
Yes, the driver files contain a lot of PV statements in plaintext, not sure if it is safe to try them.

I could do the sniffing but my 600 CU is still dead, maybe the backup of your german firmware can help there...

altenplos, 2024-03-12:
I finally borrowed a working CU and looked at the communication between CU and Station like you said. This Interface of course is much more powerful than the regular RS232 Port :)
A distance measurement is started by sending PV,4=1 and one can do things like output the compensator readings (WA,157=3).
Nevertheless I'm still looking forward for your emulation...