robots/gdm #14: Looking for info on GDM 4400 commands for RS232 port

Jesse-EE, 2024-04-27:
I've wired up a cable that plumbs my Geodimeter 4400 Robotic station to both provide power and bi directional RS232 connections.

I'm quite sure I got it right, both data lines are idling around -13v.

The 4400 seems to detect when the RS232 is plugged in, and just says "Info 23" over and over with beeps. (Timeout.)

So it's like it detects that a geodat is plugged in by looking at the idle negative voltage on the RS232 data line, then shuts down the keypad and lets the geodat take over.

However, I see nothing on my terminal program on my computer, and it does not seem to respond at all to keystrokes.

I tried 9600 and 1200 bps and neither did anything.

It's like the 4400 is waiting for a magic sequence from the Geodat (which I haven't got.)

Any information on the bitrate and/or the available commands would be greatly appreciated!

robots, 2024-04-28:
Hi,

rs232 connector is not really rs232 connector! The pinout is a bit different. There are 2 pins for 12V, 3pins for "geodimeter interface" and 2 pins serial port.
But it is not compatible with RS232 dsub9 pinout!

[image]

Here is the pinout from the iface board. Pins 1,7,5 need to be disconnected. If you touch them, firmware wants to talk over "geodimeter iface" and it will fail.

Jesse-EE, 2024-04-28:
> Hi,
> 
> rs232 connector is not really rs232 connector! The pinout is a bit different. There are 2 pins for 12V, 3pins for "geodimeter interface" and 2 pins serial port. But it is not compatible with RS232 dsub9 pinout!
> 
...

> Here is the pinout from the iface board. Pins 1,7,5 need to be disconnected. If you touch them, firmware wants to talk over "geodimeter iface" and it will fail.

Wow thanks! Are there any other clues you can drop? In general, is the "Geodimeter interface" even an async setup like RS232 is, with a start bit, some data bits, and a stop bit? Does it just use weird  bitrates? What bitrates? Any clues that are handy would be appreciated!

(I'm a firmware coder and PCB designer so I don't mind exploring weird bitrates or even weird data protocols.)

Although mine is robotic, it I cannot for the life of me find any option to tell it to auto-lock onto an active target. I do have an active target for it. 

But the old radio compartment is so huge on this thing if I could make it autotrack an active target I could stuff quite a bit of battery into where the old 450Mhz radio used to go, then use some modern (and legal LOL) serial over wireless solution.

robots, 2024-04-28:
"geodimeter interface" is very different from UART. It uses 3 pins, one clock, one data tx, one data rx. It is the same inteface they use to communicate between boards (angle, servo, edm, tracker to main board, also geodat uses this interface). If you look at this repository's files, you will find board and firmware and software to communicate over the "geodimeter interface".

In the very recent update i have tested communication directly with the "angle board".

The original idea was that the interface board (which contains radio module) could be used as internal radio for the station.

Jesse-EE, 2024-04-28:
> "geodimeter interface" is very different from UART. It uses 3 pins, one clock, one data tx, one data rx. It is the same inteface they use to communicate between boards (angle, servo, edm, tracker to [...]

That makes sense! I've worked with clock+rx+tx synchronous protocols before!

Jesse-EE, 2024-04-29:
> Hi,
> 
> rs232 connector is not really rs232 connector! The pinout is a bit different. There are 2 pins for 12V, 3pins for "geodimeter interface" and 2 pins serial port. But it is not compatible with RS232 dsub9 pinout!
> 
> ![iface](https://private-user-images.githubusercontent.com/204234/326239942-8a67aabf-b804-497f-9655-16047303ae06.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzNjc5NDgsIm5iZiI6MTcxNDM2NzY0OCwicGF0aCI6Ii8yMDQyMzQvMzI2MjM5OTQyLThhNjdhYWJmLWI4MDQtNDk3Zi05NjU1LTE2MDQ3MzAzYWUwNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyOVQwNTE0MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZTg3MmZkM2M0OTY5NjhlMzM3MjUzYjhiN2RhNTczMjhlNmMxYmUwZjk3ZTc4N2EzMDhkZjUwODliNGI2YzQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fTs46l__5WPy3joyGxwCCNxB8ArRRo_nC-3wORqxgpw)
> 
> Here is the pinout from the iface board. Pins 1,7,5 need to be disconnected. If you touch them, firmware wants to talk over "geodimeter iface" and it will fail.

If you're interested, PIN 8 has a function too:

On the Geodimeter 140 and my Geodimeter 4400, PIN 8 on the 9 pin power connector is the remote A/M button input.
It's internally pulled to 5v, and you short it momentarily to ground to activate the A/M button.

This allows you to get the cross hairs lined up so perfect then hit the A/M button without touching or disturbing the station!