robots/gdm #20: Looking for English CU

robots, 2025-02-03:
Does anyone have 600 CU with working firmware and english language ?

Maybe @bdm310  @Jesse-EE?

bdm310, 2025-02-03:
I have the 600 unit shown in my disassembly doc, working and English. I need to get either a custom ROM to dump the image or find an exploit to allow dumping without modification.

Jesse-EE, 2025-02-03:
> Does anyone have 600 CU with working firmware and english language ?
> 
> Maybe [@bdm310](https://github.com/bdm310) [@Jesse-EE](https://github.com/Jesse-EE)?

I have something like that. It is working, it is in English. It does work on 600 series stations.

However, it has had the back cover on it replaced, and the label is peeling as if maybe they moved the label from the old back plate to the new back plate. Or maybe they didn't.

The label says "CU ALPHA 671 202 130" and there's a serial number. Made in Sweden.

I have no way to read out the firmware easily. I've thought about connecting up a high channel count logic analyzer to the RAM chip but haven't done that yet.

robots, 2025-02-06:
That's why i am thinking about sending you guys "dump" eeprom. I will have it next week, so we can figure something out.

bdm310, 2025-02-06:
Ah, I just put in a Digikey order for other stuff, I should have put a few EPROMs on it. I can source my own hardware, providing a binary is certainly easier on you.

Assuming you're located in Europe and international post is annoying, I can send Jesse one as well.

Jesse-EE, 2025-02-06:
> Ah, I just put in a Digikey order for other stuff, I should have put a few EPROMs on it. I can source my own hardware, providing a binary is certainly easier on you.
> 
> Assuming you're located in Europe and international post is annoying, I can send Jesse one as well.

I live in USA...

robots, 2025-02-07:
You need to find correct chip. I dont remember the marking, but it needs to be 32k in size.

This rom is from older CU, it does not fully support the newest one. But it can "dump the variables" and it can boot the CU. Just careful :)

You need to turn on the CU using combination of keys. I dont really remember which combo. (PRG+PWR, MNU+PWR, F+PWR, ENT+PWR). Dont get too excitedy by trying various functions. If you enter "Loader" ram gets erased (end of game). You are looking for "Variable dump".

Connect to serial port, and capture the data. I think its 9600. Hit Variable dump, and wait. Its 1MB, 3bytes per byte (so 3MB) over 9600bps. Go make yourself coffee :)

Postprocessing includes removing first few bytes from dump (its junk) and running parsedump.py to convert it to binary form.

[dumper.zip](https://github.com/user-attachments/files/18705810/dumper.zip)

bdm310, 2025-02-07:
Full part number on the EPROM in mine is MX 27C256QC-12 in a 32PLCC package. 120ns access time. Any 27C256 120ns or better should do.

https://www.digikey.com/en/products/filter/memory/774?s=N4IgjCBcpgTAnBaIDGUBmBDANgZwKYA0IA9lANrgAcVALAMxUjFg0Bs8tz1VHA7N1a94TALrEADgBcoIAMpSATgEsAdgHMQAX2Kww8WFFBpIWPEVIVwsAAwACVbm60ArA6fE%2B9xyHEhpsgCqqspSAPLoALL4mLgAror42sQAtIbIJkpxFmSQlC6%2BWjog6ZSwfCiwLmy%2BxDXIygAmsilgNhCSMpAg3FIAnhJJ3bFoRUA

I'll get a few of the 70ns Atmel ones. @Jesse-EE , happy to send something to you after I make sure it works if you don't have programming tools.

bdm310, 2025-02-27:
I'm unable to power on my CU when it is not attached to the instrument. Spent some time trying to diagnose why but it's an 8 layer board and I don't want to risk shorting anything.

The PWR+ combinations you list don't do anything with the stock EPROM installed (didn't try CON). I'm simulating the PWR press by applying power to the interface pads on the back of the CU while holding down the other buttons.

I also don't have a "<-" key? I have "-/.", which doesn't do anything.

The next step would be to put the dumper EPROM in and see what happens. Do you see any issue with this? I would be connecting serial to the port on the side of the CU, holding down a button, and turning power on to the pads on the back of the CU.

bdm310, 2025-03-22:
No luck with the dumper EPROM, same behavior. Might try to make a firmware image that just initializes the serial port and dumps everything on power up. That may take a while.

bdm310, 2025-04-14:
Took a chance on an untested 610S and got lucky, unit seems to fully work and has a tracker and radio. I'll try and carefully pull dumps over this week from the CU and base. This CU does turn on by itself so there's definitely something wrong with my other one.

bdm310, 2025-04-19:
CU dump looks incorrect. Log file attached, mostly 0x55 except for a little bit at the beginning..

[control.txt](https://github.com/user-attachments/files/19823010/control.txt)

robots, 2025-04-23:
You did the wrong dump i think.(or i sent you wrong dumper rom :) )

There are two: Imem-dump and variable dump. Imem will dump the end of the memory, where recorded points are stored. Variable dump will dump from beginning of the memory. Variable dump was modified to continue until end of memory space.

bdm310, 2025-04-23:
I used <- + PWR. It immediately boots to "Dumping...." and spits that log file out of the serial port.

robots, 2025-04-24:
the dump contains 618-08 in last line, and the dumper ROM should be 648-02

robots, 2025-04-24:
I will get working cu, and i will test it again. Its been few years since i did this :)

robots, 2025-07-23:
Sorry for delay, the dump it PRG + PWR. You need to hold PRG, press PWR briefly and it will start the dump.

CU will write 'Dumping! wait' and serial will output the dump. Be sure to have recording enabled :)

Dump will not work on empty CU, so you cant practice on empty one.

bdm310, 2025-07-24:
Success this time! The end of the dump reports 618-08 but the text displayed on the screen is 648-02.00. Dumped twice, two bytes different between them and also a different final response, likely something time related. I'll pull RAM from the other control I have tonight.

https://github.com/bdm310/Total-Station-Firmware/tree/main/control/RAM%20Dumps/610S%20Control

robots, 2025-07-24:
Good to hear :-)

dont worry about the text at the end. It will report version of the dump rom, so its not interestig.

robots, 2025-07-24:
If you want to upload it , you need to run
```
 python parsedump.py dump2.txt fw.bin
```
This will convert the dump into binary format. It contains everthing, even the ROM image at the end.

To upload I have created script to assist.
```
 python upload.py fw.bin /dev/ttyUSB0
```
It will guide you through the upload process.

I can upload orange CUs. Grey CUs are not responsive. You should be able to turn it on by holding down MNU and pressing PWR briefly. On orange CUs it works, brings you to "load menu" Where you select upload, and the script will do its thing.

On grey cu the rom will just freeze in various states - Giberish on LCD, beeping, just freeze ...

bdm310, 2025-07-26:
I will be unable to pull my other CU as it does not power on in local mode and applying 12V via the back pins doesn't work with the PRG key to get in to the dumper.

TJSteed, 2026-04-05:
Rwtsv2.0@gmail.com

Jesse-EE, 2026-04-05:
> I'll get a few of the 70ns Atmel ones. [@Jesse-EE](https://github.com/Jesse-EE) , happy to send something to you after I make sure it works if you don't have programming tools.

Oh wow thank you for the kind offer! For some reason I did not get a notification when you sent it, I just now noticed it!

I have two orange CUs and one still works.

I'm pretty sure the other one used to work but now I can't get it to work with any of my 600's even though it shows stuff on the display.

Do those go blank if their battery dies too?