robots/gdm #21: Geodat 500 Data Recording Unit Serial Connection

Retro-Fitt, 2025-04-04:
While i was browsing local flea market i came across a Geodimeter Geodat 500 Data Recording Unit. I had no idea what it is for and never heard of geodimeter before. I am an electronics hobbyist and i started to play with it.

[image]

[image]

I read some of documents on this repo it seems we can communicate with it via RS-232 (5v levels?) but it only came with this cable:

[image]

which i assume just for charging internal battery and not for communicate with PC. It have a reverse polarity protection diode and 56k ohm resistor for some reason.

[image]

I tried to reverse engineer the board from N80C32 MCU but couldn't identify TX and RX pins. I could only map DB15 PINS in PCB:

[image]

Whole board:

[image]

[image]

Does anyone know TX and RX pins of this data recorder? I know in todays standard this memory is obsolete but i like playing with old junk.

Some parts on board:
N80C32 : CHMOS SINGLE-CHIP 8-BIT MICROCONTROLLER
DS1211S: Converts full CMOS RAMs into nonvolatile memories
8L05A: 5V Regulator (2 PIECES ON BOARD)
M27C256B: 256 Kbit EPROM for MCU memory (N80C32 doesn't have internal eprom)
Goldstar GM76C256ALL-70: 256K CMOS SRAM (2 PIECES ON BOARD)
LM311D: Differential Comparator 
PC74HC573T: 8-bit shift register(2 PIECES ON BOARD)
PC74HC00T: 8-bit shift register

bdm310, 2025-06-20:
If you dump the firmware, feel free to submit a PR on my firmware repo.

Based on your chip list, the 80C32 RX/TX should be routed directly out of the unit, no? I guess there's no level shifter so it's not RS232? It is possible this uses the GDM interface that is described in the wiki instead. The Ghidra disassemblies in my firmware repo have some of those routines described, probably exactly the same code was used here if that's the case.

If I'm honest, I don't think this device has much use in the modern era. I'm intending to put together an ESP32 or other shim device that provides a useful user interface and data collection system to bypass all the old Geodimeter interfaces and control units. This box is just external memory to store more data points, not super necessary with today's memory densities.

robots, 2025-07-28:
I also have this unit. There are 2 interfaces. GDM interface and Serial interface. The serial interface can be used to readout the memory - stored data. GDM interface is used normally when you plug it into the station.

The battery is lithium-thionyl primary battery. Do not try to recharge it!

Serial port stuff is in the firmware, but there was something funny about the serial connections to the unit. I have not invested lot of time to reverse engineer the schematic. 

BUT

The connector is the same as on 4000RPU. And the 4000RPU unit has multiple modes of the connector, depending on configuration of 3 pins. It can act as "RPU iface", "GDM iface", or serial port. For serial port pins 10, 12, 14 needed to be shorted. But i have no more information :)
Pin 9 is GND, pin 11 is 5V from device, pin 10 is 12V power, pins 12,13,14 select mode. I have no idea what pin 15is. Pins 1-8are the communication pins. Depends on which interface is selected.

So i would guess there is some kind of multiplexer on the board. Geodimeter liked to make schematics VERY complex.